MasterAI-Inc/libauto #1: Rewrite the `console_ui`

acu192, 2018-11-11:
It uses `pygame`, which I feel isn't the right solution. We should consider using a more full-featured GUI framework. Also last I checked, there's a bug in `pygame` (I think it's in pygame...) that prevents us from using the LCD touch screen (the coordinates of each touch were being reported totally wrongly). So, we should ditch pygame and use something that is more feature-full and that supports the touch screen.

Also! We should include a menu system in the `console_ui` that allows the user to use the push buttons to do the following things:
 - connect to Wifi
 - disconnect from wifi
 - see the car's IP address
 - calibrate the car
 - run built-in programs as demos
- read sensors... (that's probably just a sample program)

Final note: The new `console_ui` should keep the same RPC server interface so that the other programs don't have to change.

acu192, 2018-11-12:
For the menu, here are the commands to query for and view the nearby wifi:
```bash
> nmcli device wifi rescan
> nmcli device wifi list
```

acu192, 2019-05-22:
Regarding the menu system, _some_ progress is made in 3af6ae5b57fe732982cc0b27c98c76b6b6b1dff6. That commit created a background process to monitor the buttons, looking for a certain combo of buttons, with the idea that combo of buttons will launch the on-screen menu system. Currently, however, it doesn't launch the menu system (there is no menu system); it instead just runs the calibration routine.

acu192, 2019-05-22:
Maybe there's a menu option to make the car be a wifi hotspot? That would be cool. It seems easy enough with the `nmcli` utility. [docs here](https://developer.gnome.org/NetworkManager/stable/nmcli.html)

Hariselvan04, 2022-05-10:
my autoauto car is no starting with ui it shows ERROR: delete swap file

acu192, 2022-05-10:
> my autoauto car is no starting with ui it shows ERROR: delete swap file

@Hariselvan04 Please fill out the contact form here: https://www.autoauto.ai/contact/ and then the AutoAuto team can assist you from there.