MasterAI-Inc/libauto #24: Make backend device "pluggable"

acu192, 2019-08-20:
Each backend should offer a list of "components" which can be acquired and released. Some components are enabled by default. Some components _must_ remain enabled (no disabling). Some components are exclusively held (new connections will close the old ones), while other support multi-connections. This should all be pluggable.

As such, we could run the library on other devices.

Or we could make a mock device so we can run libauto on a desktop computer.

glenwinters, 2019-10-29:
Hey @acu192. At the open source meetup, we talked about how you only test your code with a physical car right now. I'm interested in helping setting up a suite of tests that can run without any hardware. I think this issue would help achieve that.

Do you have some suggestions for approaching this?

acu192, 2020-01-08:
This is done in v2.x!

closed 2020-01-08 by acu192